wchill/SwitchInputEmulator #1: Is this based on what I did?

leo60228, 2018-09-12:
I made something almost exactly like this (except without the TwitchPlays aspect) all the way back in January, and posted a video to ReSwitched. I didn't finish it before my HDD died, so it's not open source (if you want, I can link the video, but I don't have it right now). It's fine right now, but if you got the idea from my video, could you credit me in the README?

wchill, 2018-09-12:
This is more or less an original idea (well, as original as a Twitch Plays can be). All code except parts of the base controller code are completely original, and even the controller code has been heavily modified. I didn't actually see any such video in RS so I have no idea what you're referring to specifically...

leo60228, 2018-09-12:
Okay. I'll try to find it later. The video is from January; sorry if that was confusing.

wchill, 2018-09-22:
Closing for now

closed 2018-09-22 by wchill