wchill/SwitchInputEmulator #4: Serial communication from arduino sketch?

jwhop, 2019-03-11:
Hi, i had a quick question about your project. I hope to use it to remap certain buttons of a gamecube controller. I currently have an arduino sketch reading the gc input through a data pin and writing it to serial. I have the baud rate matching the one in the joystick.c file, but nothing seems to be happening once i flash the hex file onto my arduino (uno r3). Is this project designed for serial communication from the same board? I am not sure if my issue is sending the right bytes out from my sketch or if im not setting up communication properly between the firmware and the sketch. I am new to all things arduino firmware /HID related so sorry if the question is poorly phrased. Thanks in advance!

wchill, 2019-03-11:
Hi @jwhop,

If you flash the sketch before you flash the hex file, then I'm guessing it should be possible? It is definitely not something I'd consider a supported scenario since it is very much a hack, but as I understand it the ATmega328p's TX is connected to the 16u2's RX and vice versa, so it sounds like it should work (though I haven't tested it).

I would highly recommend you buy a second Arduino board or a clone if you can though just for ease of debugging and whatnot.

jwhop, 2019-03-15:
Got it working! Ill close the thread

closed 2019-03-15 by jwhop