wchill/SwitchInputEmulator #8: Sending controller inputs via arduino instead of python?

TeslaBolt, 2019-03-31:
(First off, sorry if this isn't the correct place to ask this)
I am trying to create a controller for the switch, and ideally i'd want the same arduino checking buttons and communicating with the switch (this is after I get this step working though).

I have successfully used the client.py script to communicate and send commands to the switch, so I thought that if I recreate the script in arduino, it should work for my purpose.
(Instead of PC -> Serial Adapter -> Arduino (Joystick.hex) -> Switch, i'm trying to use
Arduino (running script) -> Arduino (Joystick.hex) -> Switch 

Well, I've done that (except for the testbench functions) and it doesn't work (I can't tell exactly why, but i'm pretty sure it's just not syncing).
Link to arduino code: https://pastebin.com/XWmZmS7y

Is there another way to approach this, or are there any mistakes in my code that is causing it to not work? (I am new to arduino programming and serial communication stuff)

Thank you, and again I apologize if this isn't the right place for this.

wchill, 2019-04-01:
I don't see anything that *looks* incorrect from a first glance, but it's hard to say. Unfortunately due to work/moving I don't have a lot of time to debug this.

What I would recommend is trying to compile the emulator emulator (yeah, I know, sounds funny) - if you install socat, run make and then make run-emu, it runs the same code that would run on the Arduino as a binary using emulated serial ports via socat. See https://github.com/wchill/SwitchInputEmulator/blob/397c14977dac8f3094f093852b44af5a47ae4559/Arduino/Makefile#L68 for what I mean.

You can try piping the Arduino's (the one running the script) serial input/output to this file and then see what's going on by inspecting the traffic. If you need more help with this, I can answer questions for issues you might run into.

TeslaBolt, 2019-04-06:
Hi, I've recorded some communications between the arduino and the emulator, and i've found something interesting.
When connected to the emulator, it syncs and it works.
However, when connected to the other arduino (the joystick.hex one), it doesn't work (it doesn't sync).
I know the joystick.hex arduino works though, as I can send commands through the computer.

Here's a log from the arduino to the emulator (I used interceptty to log the communication):
https://pastebin.com/Cv0D3TXH

Any ideas on why this might be happening?

wchill, 2019-04-07:
Just want to make sure, you have the Arduinos wired up correctly right? TX->RX and RX->TX.

TeslaBolt, 2019-04-08:
Yep

javmarina, 2019-05-01:
I have a similar problem. I have compiled Joystick.hex and flashed it to my Arduino Uno R3. The device is properly recognized as a gamepad and it sends the default controller state (centered sticks and no buttons pressed). I'm using a USB to Serial adapter (TTL, FTDI FT232RL chip) and wiring it as follows:
- USB port connected to computer
- Vcc and GND connected to 5V and GND in Arduino board
- TX in adapter to RX in Arduino (pin 0)
- RX in adapter to TX in Arduino (pin 1)
The adapter is recognized as a serial (COM) port. However, even if I send bytes (using the provided script or by other means) the Arduino doesn't send back responses. The adapter has a LED that lights up when it's transmitting to the Arduino and I can confirm that data is being sent to the board. However, the LEDs in the Arduino board are always off and the computer doesn't get a response back.
My understanding of serial communications in Arduino boards (AVR) is limited, but I have read that TX and RX have to do with the USB B plug. As far as I know, the USB has to be used to connect with the USB A ports in the Switch dock. Do you know what might be happening?

wchill, 2019-05-01:
The Arduino flashed with Joystick.hex effectively bypasses the ATMega328p on the board and uses only the ATMega16U2 (the microcontroller handling USB comms). That should always be plugged into a device.

Having said that, I realize now that I might have gotten the wiring order wrong when I described it here. The TX/RX pins on the Arduino are relative to the ATMega328p. So the ATMega16U2's RX is connected to the ATMega328p's TX pin (as labeled on the board) and vice versa.

So you should probably try connecting the adapter's TX to the Arduino's TX and RX to RX.

javmarina, 2019-05-02:
Thank you so much. It's working now for me.
Consider writing a reminder about this aspect, as it's not very trivial.

wchill, 2019-05-03:
@TeslaBolt see above for fixed wiring instructions. My bad.

I'll keep this issue open until I get around to fixing the README.

TeslaBolt, 2019-07-06:
@wchill Thanks for the notice about the wiring, however I tried that and it still doesn't work.

The joystick flashed arduino does get recognized my computer so that isn't an issue, so I guess the issue lies in my 2nd arduino (the one sending the serial from my rewriting of the code), or the code itself, even though my 2nd arduino does work correctly when connected to the emulator. I guess there's something not communicating right between the 2 arduinos specifically.

shendriksza, 2019-08-12:
@TeslaBolt I want to try something similar, did you ever get this to work? I'm going to try your code