wchill/SwitchInputEmulator #10: Arduino board is not recognized by Switch

javmarina, 2019-05-04:
I have compiled the Joystick.hex and written a script in order to communicate via serial. Everything is wired correctly, as the Arduino UNO R3 receives the packets correctly. When synced, the board answers with 0x90 and if I connect it to the computer and go to https://html5gamepad.com/, it is shown as a gamepad and the button presses are correct. So the issue has to do with the Switch not recognizing it as a valid controller.
Surprisingly, if I modify some parameters in the Descriptors.h file, it works. After a lot of trial and error, I have a working Descriptors.h. The modified version is inspired by the original from https://github.com/progmem/Switch-Fightstick. However, there are only a few differences between my modified version and the original file in this repository:
- .ProductID was 0x00C1, now it's 0x0092
- .ReleaseNumber was VERSION_BCD(5,7,2), now it's VERSION_BCD(1,0,0)
- .MaxPowerConsumption was USB_CONFIG_POWER_MA(250), now it's 500 mA
- ProductString was USB_STRING_DESCRIPTOR(L"HORIPAD S"), now it's USB_STRING_DESCRIPTOR(L"POKKEN CONTROLLER")

With these four changes, the Switch now recognizes the Arduino board. Note that I tried pressing L+R with the original file with no luck.

wchill, 2019-05-06:
Hmm, something doesn't seem right here. The HORIPAD S descriptors are from an actual working controller, and in theory any USB HID compatible device should be recognized properly. I've been using those descriptors for months with no issues.

javmarina, 2019-05-06:
I know it's strange. I may have done something wrong, but I followed the same steps for both descriptors. The board was synced and sending 0x80 ACKs.

wchill, 2019-05-06:
A bit of a long shot, but do you have a setup that would allow you to MITM USB traffic? If so, I'd be interested in seeing what's going on.

javmarina, 2019-05-06:
For now, I don't know how to analyze USB traffic. I will try it someday when I have spare time. I will also try modifying some of the parameters and install again the original firmware, just to check if I did something wrong. My plan is to start with the original code and then modify one of the parameters until it works again.

wchill, 2019-05-06:
If you get a logic analyzer (I have a Saleae Logic Pro 8, but there are cheaper analyzers available) and a USB extension cable, I had some luck cutting into the cable and attaching my logic analyzer probes to the D+/D- wires. Since we are doing communication at USB 1.1 speeds, the sample rate requirement is not that high if you go this route.

javmarina, 2019-07-02:
I just tried flashing the original firmware again and I got it working. My guess is that I didn't configure the system correctly. I tried connecting the cables in different order and it worked.
Btw, do you know if those changes can affect something or are just "cosmetic"? I wonder if they make a real difference while playing.

closed 2019-07-02 by javmarina

wchill, 2019-07-02:
@javmarina the USB related changes are technically just cosmetic; the descriptors I use are more "correct" since they're from a real product whereas the descriptors from the other project are modified, but there isn't really any difference in functionality.