wchill/SwitchInputEmulator #12: Unable to Sync with Arduino

tgb20, 2019-05-16:
I am using an Arduino UNO Rev3 flashed with Joystick.hex and when connected to a computer and using [this](https://html5gamepad.com/) tester it shows up as a HORIPAD S with the correct settings.

When using client.py to try and communicate with the the Arduino it cannot sync and sends back seemingly random values.
Here are some values I have received back:
`0x00` `0x60` `0x26` `0x87`

I am communicating to the Arduino from a Raspberry Pi 2 B.

wanglamao, 2020-01-11:
I had similar problems. when running client.py, it tells me cant not sync and two packet error. 

In https://html5gamepad.com/, arduino can be identified as horipad, but client.py never syncs. 

I  am not sure if wiring was correct. to run client.py, I did like this: pc -> uart usb converter->arduino uno r3 -> pc.

I also tried to swap tx and rx but in this case, client.py will get stuck in wait_for_data().

Thank you in advance.

ammounce, 2020-01-18:
I'm having identical issues to wanglamao, has anyone been able to figure this out?

wanglamao, 2020-01-18:
Hi, @ammounce I made a few progress after fixing my wiring problem. So, make sure your wiring is correct. If you are using arduino uno r3, your wiring: arduino's reset -> arduino's gnd, uart rx->arduino rx, uart tx->arduino tx. However, these 3 wires may be enough or not enough. In my case, I have another connection: uart gnd->arduino gnd. 
With this wiring, I can run other similar repos and thus control the switch using my PC. But I still have problems with this repo because I found that there're still issues with receiving data from arduino which further leads to unsuccessful handshake/sync.

ammounce, 2020-01-18:
Great thanks! Which repo did you end up using?

wanglamao, 2020-01-18:
https://github.com/ebith/Switch-Fightstick
My modification is based on this reppo.