wchill/SwitchInputEmulator #13: Teensy Compilation Error

adstep, 2019-05-18:
When I compile with MCU=at90usb1286 I see the following error:

```
 [GCC]     : Compiling C file "Joystick.c"
avr-gcc -c -pipe -gdwarf-2 -g2 -mmcu=at90usb1286  -fshort-enums -fno-inline-small-functions -fpack-struct -Wall -fno-strict-aliasing -funsigned-char -funsigned-bitfields -ffunction-sections -I. -DARCH=ARCH_AVR8 -DF_CPU=16000000UL -mrelax -fno-jump-tables -x c -Os -std=gnu99 -Wstrict-prototypes -DUSE_LUFA_CONFIG_HEADER -Iinclude/  -I. -Ilufa/LUFA/.. -DARCH=ARCH_AVR8 -DBOARD=BOARD_NONE -DF_USB=16000000UL  -MMD -MP -MF obj/Joystick.d src/Joystick.c -o obj/Joystick.o
In file included from include/Joystick.h:42:0,
                 from src/Joystick.c:21:
include/avr.h: In function ‘USART_Init’:
include/avr.h:27:5: error: ‘UCSR1D’ undeclared (first use in this function)
     UCSR1D = 0;                         // no cts, no rts
     ^
include/avr.h:27:5: note: each undeclared identifier is reported only once for each function it appears in
lufa/LUFA/Build/DMBS/DMBS/gcc.mk:213: recipe for target 'obj/Joystick.o' failed
make: *** [obj/Joystick.o] Error 1
```

It compiles fine with MCU=atmega16u2. Anyone else get this problem before?

wchill, 2019-07-08:
The at90usb1286 support is from the original project. You'll need to consult the datasheet for that MCU in order to rewrite the serial receive code, as the registers are different.

parkerlreed, 2019-09-05:
I assume this means Teensy 3.6 is also not supported?

wchill, 2019-09-06:
Teensy 3.6 doesn't use an AVR core, it's running an ARM Cortex M4. So everything is completely different for that.

parkerlreed, 2019-09-06:
Yeah figured. Would be interesting as the 3.6 also has USB host so you could make it into a converter as well. 

Thanks.

wwboynton, 2020-08-27:
Probably worth updating the README to reflect that `2.0++` is not really supported, since newer teensy boards are not compatible. Thanks for your work on this project!