wchill/SwitchInputEmulator #16: How do I use this?

lorecast162, 2019-09-28:
The project site seems to not work anymore (gives DNS_PROBE_FINISHED_NXDOMAIN error) I don't know how to use this thing. I just programmed the .hex file into the 16u2 chip on my UNO R3.

kupokappa, 2019-10-11:
You need a USB to UART bridge. Someone else responded to an issue on this repo with the correct wiring, and there's probably some resources out there that'll tell you what to do for your specific adapter. I'm using a Silicon Labs CP210x USB to UART adapter. Wiring into an Arduino Uno R3 goes as follows (USB to UART -> Arduino):

DTR -> RESET
RXD -> RX
TXD -> TX
5V -> 5V
CTS -> GND
GND -> GND

From there, you can just use the example client.py.

INobody95, 2020-03-16:
Hello,
after flashing the Arduino while waiting for my ordered CP2102 USB to UART adapter I honestly got not idea what to to next? Wiring from your post above and then? What file do I have to open (client.py in console?) and what to plug in where?
Native Arduino USB into the switch dock, USBtoUART into PC, Controller into PC?

danvx6, 2020-12-25:
> You need a USB to UART bridge. Someone else responded to an issue on this repo with the correct wiring, and there's probably some resources out there that'll tell you what to do for your specific [...]

I think I may have bought the wrong kind of UART adapter, my pins are labeled:
3V3
GND
TXD
RXD
GND
5V

This is the adapter in question: [https://www.amazon.co.jp/-/en/gp/product/B072K3Z3TL/ref=ppx_od_dt_b_asin_title_s01?ie=UTF8&psc=1](url)

It does not have DTR or CTS pins. Will it still be usable for this project?

wchill, 2021-01-04:
DTR and CTS are not needed. What you have is fine, just ignore the pins you don't have

closed 2021-01-04 by wchill