wchill/SwitchInputEmulator #17: Change baud rate

javmarina, 2019-10-13:
I have seen that the baud rate is set to 19200 by default. Is there a reason why this value is used? If not, I suppose it's possible to increase it as long as the same value is used in the Joystick.c and the PC side, plus the USB-to-serial adapter and the board must support that value. What values do you think are valid?

wchill, 2019-10-13:
9600 is too slow and faster speeds are not necessarily better - you are going to be limited by the 125Hz USB poll rate in most cases anyway.

But you can try higher and see if it helps your specific application.

javmarina, 2019-10-18:
I finally got it working with a baud rate of 1000000 bps (1 Mbps). For future reference, and anyone interested, you first have to figure out the available baud rates for your microcontroller.

In my case, I was using an Arduino Uno R3, which has a ATmega16U2 chip. Looking at the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/doc7799.pdf), table 18-1, you can see the formula that is used to compute the UBRR register value. Note that this firmware uses Asynchronous Normal mode (U2Xn = 0), so the formula in my case was `UBRR = (freq/(16*baud))-1`. The oscillator frequency is defined in the Makefile (`F_CPU`), which is 16 MHz by default. So, the maximum baud rate is 1 Mbps (see table 18-12 for reference), when UBRR=0.

This firmware uses the default LUFA formula (taken from `lufa/LUFA/Drivers/Peripheral/AVR8/Serial_AVR8.h`) for computing the UBRR register for a given baud rate. This macro (`SERIAL_UBBRVAL(Baud)`) is defined in `include/avr.h` and is only used in the definition of the `USART_Init()` function. Note that this formula differs slightly from the one in the datasheet. The LUFA formula always gives +0.5 value as a result. This is because it was found that some register values are not correctly computed if using the Atmel formula). By taking advantage of integer division logic, the LUFA formula fixes those issues. See [here](https://forum.arduino.cc/index.php?topic=46422.0) and [here](https://www.nongnu.org/avr-libc/user-manual/FAQ.html#faq_wrong_baud_rate) for more information.

For a baud rate of 1000000, both formulas give 0 after rounding. 0 is the value for a baud rate of 1M, as defined in the table 18-12 of the datasheet.

Then, you have to change the argument type of the `USART_Init()` function from int to uint32_t, because such high values would cause an overflow otherwise. Modifications are needed in `include/Joystick.h`, `include/avr.h` and `include/avr_mock.h`. Then, change the argument of `USART_Init()` in the `SetupHardware()` function.

Finally, you have to make sure that your USB-to-serial adapter can handle the desired baud rate. I'm using a FT232R chip, so [here](https://www.ftdichip.com/Support/Documents/AppNotes/AN232B-05_BaudRates.pdf) are the available values (see section 1.3). Then, change the baud rate port setting in the PC side and you're set!

closed 2019-10-18 by javmarina