wchill/SwitchInputEmulator #25: Speedups

wwboynton, 2020-09-01:
Hi, there!

I'm looking to time-optimize this/similar code by speeding up some of the inputs, or even just produce an example that will mash the A button. However, I'm new to the world of C/C++, and I haven't used LUFA before either. I've tried reading around in the [related](https://github.com/shinyquagsire23/Switch-Fightstick) [repos](https://github.com/kidGodzilla/woff-grinder) but it seems like all of these repos have tremendous delay between every single input. 

Here's what I've tried already:

- I've tried adjusting the duration of the button presses in `static const command step[]`, but it seems like those relate to how long the button is held, rather than the interval between inputs.
- I've tried adjusting the `.PollingIntervalMS` in `Descriptors.c` but that doesn't seem to be the right place to look.
- It seems like the `_delay_ms(250);` in `Joystick.c` only gets used at the end of a whole sequence.

Can you help point me in the right direction?

wchill, 2020-09-01:
Hi @wwboynton, while I haven't worked on this in a while, maybe I can point you in the right direction.

How much delay are you looking at here? There are some inherent lower limits to the latency that are not trivially resolvable, but if you're seeing a large amount of delay per set of inputs then that's something that can probably be addressed.

wwboynton, 2020-09-01:
Hi, sorry. It seems my problem was totally not what I thought it was. Thanks for responding to a barely-related issue on a repo you haven't worked on in so long. I'll detail my issue below for anyone who happens to stumble upon this down the road:

> It turns out that by failing to send a gap between my presses of the A-button, it was sending them more frequently than the polling rate and effectively just registering in the system as one long press. By adding some deliberate batches of nothing between my presses, I got the desired result with no issue.

closed 2020-09-01 by wwboynton

wchill, 2020-09-02:
Yep, for future reference, the max number of updates is 125Hz (USB 1.1 standard). That's the reason why the ATMega is sending back 0x90 every 8ms, so you can synchronize your code around it.