wchill/SwitchInputEmulator #26: Does not compile on Windows

danvx6, 2020-12-24:
I'm unable to compile on Windows 10, which seems to be an issue with the library sys/select.h not existing on Windows. Is there an alternative?

`In file included from include/Joystick.h:40,
                 from src/Joystick.c:21:
include/avr_mock.h:9:10: fatal error: sys/select.h: No such file or directory
    9 | #include <sys/select.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:62: emulator_obj/Joystick.o] Error 1`

sisco0, 2020-12-25:
Have a good day @danvx6 , the alternative is based on using `winsock2.h` instead of `sys/select.h`.

danvx6, 2020-12-25:
Thanks for the hint @sisco0.

I tried substituting sys/select.h for winsock2.h but that creates more errors because the functions inside are different. After a google search I found suggestion to change select() to FlushFileBuffers(), but that doesn't seem to work in this case and leaves other errors:
<code>`c:\mingw\include\winbase.h:1610:42: note: expected 'HANDLE' {aka 'void *'} but argument is of type 'int'
 1610 | WINBASEAPI BOOL WINAPI FlushFileBuffers (HANDLE);
      |                                          ^~~~~~
gcc -o emulator emulator_obj/Joystick.o
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: emulator_obj/Joystick.o:Joystick.c:(.text+0x4f): undefined reference to `__WSAFDIsSet@8'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: emulator_obj/Joystick.o:Joystick.c:(.text+0xb4): undefined reference to `select@20'
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:58: emulator] Error 1`
</code>

At this point I may just try to compile in a linux vm since tbh I don't understand what is going on in the program enough to replace these functions with ones available to Windows.

sisco0, 2020-12-25:
Have a good day, you are obtaining linker errors, you should use -lws2_32 as a linking parameter.
Even more, `select` function is the correct for both headers so no `FlushFileBuffers` function should be used.

sisco0, 2020-12-25:
By the way, I recommend you using _WSL 2_ (Windows Subsystem for Linux) feature of _Windows_ in order to accomplish your task. It is a very convenient approach to integrate _Linux_ under your _Windows_ environment.

danvx6, 2020-12-26:
Thanks for all the help. I meant I used FlushFileBuffers in place of fsync, my mistake.
After following your advice I was able to successfully compile Joystick.hex! Thank you!

closed 2020-12-26 by danvx6